osexpert/ccs32clara-chademo #2: Firmware suitable for "old" hardware?

nogi47, 2025-06-22:
According to Dongguan there are two versions of the adapter with  different IO port configurations.

Does your firmware work with both versions or is it strictly for the latest version?

Another question:
Since I'm the guy who proposed to calculate the actual battery voltage using SOC and the difference between min and target voltage
in the German Going Electric forum, I checked your function GetEstimatedBatteryVoltage in cademoCharger.cpp.

I wonder why you are using an offset of 117**point**2 volt.
Is there a reason for the .2 ?

Please keep up the very good work.

-NoGi

osexpert, 2025-06-22:
Hi,

"Does your firmware work with both versions or is it strictly for the latest version?"
I have the latest version adapter so I assume it won't work with the older hw version. If its only a matter of different GPIO's I guess it could be possible to make it work. But I would need access to the old version adapter or collaboration with owner willing to risk bricking it (not likely, but always a chance). Maybe it is possible to detect hardware version from data in flash, in case the same firmware binary can be used. But I don't know.

"Is there a reason for the .2 ?"
From the 2 known "facts" I used:
Leaf: targetVoltage = 410V, nomVolt = 355V
iMiEV: targetVoltage = 360V, nomVolt = 326V
355 = a * 410 + b
326 = a * 360 + b
Solved: a=0.58, b=117.2

Some claim Leaf has nomVolt of 350, some claim it is different between Leaf models. But as an estimate, I just had to choose some numbers and stick with it:-)

"Please keep up the very good work."
Thanks:-)

osexpert, 2025-06-24:
I found the Going Electric forum you mentioned and I saw you tested a few chargers and it seems to work even with older hardware. That is a surprise, but really great. Thank you for testing.

luipir, 2025-06-29:
@nogi47 can you share link to Boing Electric thread please? Probably I've a similar hw (branded ev-boy) but, if it is the same Dongguan product, no idea if it is old or new hw.

uhi22, 2025-06-30:
https://www.goingelectric.de/forum/viewtopic.php?p=2344926#p2344926
I think it would make sense to create a hardware-versions.md with some pictures of the tested hardware, to avoid confusion.

osexpert, 2025-06-30:
Found online where the buttons can be seen: https://evniculus.eu/cdn/shop/files/IMG_4238.jpg
If you order many adapters from Longod Dongguan they will probably brand them in what brand you like, as they are sold in at least 4 different brands already:-)
About the hardware versions, I saw a difference in the logs:
Mine:
QCA7000 sig is 0x0
QCA7000 sig is 0xAA55
In log from @nogi47:
QCA7000 sig is 0xFFFF
QCA7000 sig is 0xAA55

So if @nogi47 has the older hardware version, I am guessing it uses a different ccs module or is wired slightly different (MISO pulled high instead of floating?), but still seem to be working.

luipir, 2025-06-30:
> Found online where the buttons can be seen: https://evniculus.eu/cdn/shop/files/IMG_4238.jpg If you order many adapters from Longod Dongguan they will probably brand them in what brand you like, as [...]

busy period this month, but I'll try to log to check if mine is the same longgood hw. Is it just serial dump from usb when connector is plugged in the car?

osexpert, 2025-06-30:
> busy period this month, but I'll try to log to check if mine is the same longgood hw. Is it just serial dump from usb when connector is plugged in the car?

Yes, logging works the same.

nogi47, 2025-07-01:
I bought my adapter  in February 2025. It is the one with a hole in the  CCS2-side stop-button  for the use of a small padlock.
It was loaded initially with firmware   V3.3.8.6.B48.9.1 My407ccs2chademo.bin.

I think I've got the new hardware version as the change to V 3.3.8xx toke place in February/March of twenty-four.

-NoGi(47)

closed 2025-07-01 by nogi47