osexpert/ccs32clara-chademo #9: V2L Discharger

Foomi84, 2025-07-19:
I tested firmware 36 with the CCS Chademo Adapter and V2L Discharger on a Nissan e-NV200.
The devices communicate with each other. The charging process begins, but stops after a few seconds.
I tested the V2L CCS Discharger with a 2025 Dacia Spring and a 2017 Tesla X P100D, and it worked.
When the devices draw power from the car, they report that charging is in progress, but the voltage is negative.
Maybe my logs will help us work on future versions.

[20250719_CCSChademo&V2L.txt](https://github.com/user-attachments/files/21327964/20250719_CCSChademo.V2L.txt)
[20250719b_CCSChademo&V2L.txt](https://github.com/user-attachments/files/21327963/20250719b_CCSChademo.V2L.txt)

[image]

osexpert, 2025-07-19:
Hi. I think chademo support discharge, so it could be possible. The firmware currently has no (deliberate) support for it, but it could be interesting to try to add support for it. Later when/if I get time, and will work on this, I can ask you to be guinea-pig?

osexpert, 2025-07-22:
"which works with their adapter" do you have this adapter? What brand is it? Url/image? Do you know if it support discharge and charging with the same firmware? 

"But the voltage is negative"
I did not see any negatives in the log. Do you see this on a display on the discharger? Is it negative when used on native ccs  too, or only with longgod  adapter?

Thanks.

Foomi84, 2025-08-12:
I'm sending logs from an attempt to use the Adapter with the CCS Disharger and from charging on a Lidl charger with two CCS plugs. These chargers also didn't work with the original software. Unfortunately, I tested with a nearly full battery.
When I use the CCS Discharger with cars with CCS2, the cars detect it as a charging session.
1. It worked in a 2017 Tesla X, which doesn't officially support discharging. Tesla recognized it as a charging session.
2. It worked in a Dacia Spring, which supports the V2L Type2 discharging adapter. When using the V2L Type2, the battery charge indicator scrolls from right to left – the car recognizes this as discharging. When I used the CCS Discharger, the charge indicator scrolls from left to right – just like during charging.

[20250812 Adapter V2L.txt](https://github.com/user-attachments/files/21733876/20250812.Adapter.V2L.txt)
[20250812 Lidl dual CCS Charger.txt](https://github.com/user-attachments/files/21733877/20250812.Lidl.dual.CCS.Charger.txt)
[20250812AdapterV2L_2try.txt](https://github.com/user-attachments/files/21733878/20250812AdapterV2L_2try.txt)

Foomi84, 2025-08-12:
The CCSV2L Discharger was created to discharge Teslas. Teslas don't support discharging, so the discharger has to fake that is a charger.

justingist, 2025-08-20:
If this works AND can get to work with I-MiEV overall, I may have to look into one of these dischargers as I also have a CCS car... having an adapter to discharge both would be a game changer for camping and power outages... going to be following this

osexpert, 2025-08-24:
@vlad-evniculus do you have pictures (or model name) of the different discharges that produced the files?

osexpert, 2025-08-25:
Hi, I have added a few experiments here:
20250825-119
https://github.com/osexpert/ccs32clara-chademo/actions/runs/17206133429
Experiment3: if detecting a discharger (discharger seem to say its output voltage is the same as target voltage), then switch to chademo discharging.
Thanks.

vlad-evniculus, 2025-08-30:
No success with the 20250825-119, I have tested only main+ longood. Will post pics later

[SaveWindows30.8.2025_г._22-47-36.TXT](https://github.com/user-attachments/files/22060991/SaveWindows30.8.2025_._22-47-36.TXT)
[SaveWindows30.8.2025_г._22-44-43.TXT](https://github.com/user-attachments/files/22060990/SaveWindows30.8.2025_._22-44-43.TXT)

osexpert, 2025-08-30:
Thanks. I see some of my experiments may be in conflict... Ill be back with a new one:-)

osexpert, 2025-08-31:
New experiment (same as in issue 18):
20250831-131
https://github.com/osexpert/ccs32clara-chademo/actions/runs/17360175112

Add a +-20volt cyclic offset to the estimated battery voltage, to try to defeat the cars voltage checks. This could hopefully allow the discharge hack running longer.

Thanks.

vlad-evniculus, 2025-09-03:
[SaveWindows3.9.2025_г._20-28-07.TXT](https://github.com/user-attachments/files/22124568/SaveWindows3.9.2025_._20-28-07.TXT)
[SaveWindows3.9.2025_г._20-25-06.TXT](https://github.com/user-attachments/files/22124567/SaveWindows3.9.2025_._20-25-06.TXT)
No success with this version as well.

These are the other tested dischargers

[image]

osexpert, 2025-09-03:
New experiment:
20250903-137
https://github.com/osexpert/ccs32clara-chademo/actions/runs/17444003632
Wildly guessing the car stop responding because of discharge negotiation during discovery. Thanks and good luck.

osexpert, 2025-10-02:
@Foomi84 this is still on Nissan e-NV200?

vlad-evniculus, 2026-02-07:
Hi, attached are 4 failed attempts for V2L with ZE0 at around 50% SOC. 

[SaveWindows7.2.2026_г._17-09-21.TXT](https://github.com/user-attachments/files/25152571/SaveWindows7.2.2026_._17-09-21.TXT)
[SaveWindows7.2.2026_г._17-06-43.TXT](https://github.com/user-attachments/files/25152568/SaveWindows7.2.2026_._17-06-43.TXT)
[SaveWindows7.2.2026_г._17-15-17.TXT](https://github.com/user-attachments/files/25152569/SaveWindows7.2.2026_._17-15-17.TXT)
[SaveWindows7.2.2026_г._17-11-44.TXT](https://github.com/user-attachments/files/25152570/SaveWindows7.2.2026_._17-11-44.TXT)

vlad-evniculus, 2026-02-08:
Hi, now it should be activated. Maybe the high SOC is a problem? 

[SaveWindows8.2.2026_г._9-07-47.TXT](https://github.com/user-attachments/files/25159658/SaveWindows8.2.2026_._9-07-47.TXT)
[SaveWindows8.2.2026_г._9-06-17.TXT](https://github.com/user-attachments/files/25159659/SaveWindows8.2.2026_._9-06-17.TXT)
[SaveWindows8.2.2026_г._9-03-28.TXT](https://github.com/user-attachments/files/25159660/SaveWindows8.2.2026_._9-03-28.TXT)
[SaveWindows8.2.2026_г._9-02-20.TXT](https://github.com/user-attachments/files/25159661/SaveWindows8.2.2026_._9-02-20.TXT)

osexpert, 2026-02-08:
3 of them fails with suddenly can not send chademo CAN messages anymore:
[09:07:21.215]IN¡û¡ô[cha] can transmit: msg:0x108 timeout
[cha] can transmit: msg:0x109 timeout

One of them get further: SaveWindows8.2.2026_?._9-03-28.TXT
But then also stop at car stopping before starting:
[09:03:17.004]IN¡û¡ô102.Status 0xc8 -> 0xd8
[cha] St[09:03:17.030]IN¡û¡ôopping before starting

If you try the latest version, the stopping before starting check is removed, but it may or may not help:
https://github.com/osexpert/ccs32clara-chademo/releases/tag/20260203-280

At least to me in 3 cases, it seems like the car chademo controller just decide it want to suddenly turn itself off. In the 4th case it decide to set a flag indicating it want to stop. So it is obviously not happy about something (I would say it seems to be malfunctioning). 

But yes, a tricky problem. The newer leafs works flawlessly, and this is what I have, so it "always work for me".

osexpert, 2026-03-15:
This issue is getting too long:-)
The last logs were on 20251217-232, and it has been a lot of changes since then.
I also tested evniculus discharger on 20260312-319 with Leaf Ze1 and it works without issues (using discharge mode).
So I think its best to close this issue and instead open a new one when the next problem appears, on latest fw version.
Thanks.