osexpert/ccs32clara-chademo #11: In Finland K-Lataus failed to charge

Kazhuu, 2025-07-22:
I want to charge in Finland at K-Lataus and the charger it seems to try to start, but then the charged indicate failing condition and it stops charging. My car is Nissan Leaf 2018 ZE1 with 40 kWh battery. Here is the logs about the session

[serial_20250722_235303.txt](https://github.com/user-attachments/files/21375524/serial_20250722_235303.txt)

I tried this with the latest master I believe. I just downloaded the binary from the CI action. The version from the logs is `ccs32clara-chademo 20250718-58`.

osexpert, 2025-07-22:
Hi. I recommended to use a release as they are tested, but  main should normally work too.

The log just say the charging is terminated by the charger itself, for unknown reason. The charger seems to try to deliver current current/amp, even before the car ask for it... I have never seen this before. It may be voilating the spec...

There is usually a model number plate on chargers, could you take a picture of it? Its nice to know what kind of charger we have here...I am guessing a completely new/unknown model.

osexpert, 2025-07-22:
Ccs2. Can evse deliver current if ev targetcurrent is 0?

ChatGPT:
No, in CCS2 DC charging (ISO 15118 or DIN SPEC 70121), if the EV sends TargetCurrent = 0, the EVSE will not (and must not) deliver current.

🔒 Why?
The EVSE is not allowed to push current unless the EV explicitly requests it. The EV is always the current controller; the EVSE only responds within its own limits.

Kazhuu, 2025-07-23:
[image]

Here is the charger. I don't know the manufacturer of this as it was not written there. I think this is also common in Finland elsewhere. I can test a bit more to see.

uhi22, 2025-07-23:
It would be helpful to log the content of the first CurrentDemandReq and response. This should give a hint why the charger terminates the session.

osexpert, 2025-07-23:
"I don't know the manufacturer of this as it was not written there" it should be written on a metal plate, typically 5x10 cm i size, near the ground. I have not yet seen a charger that does not have this, seems like it may be required by law.

osexpert, 2025-07-23:
Analysis of log:

23:30:09.951 Ccs stop reason 0x4. Power off pending...

0x4 is STOP_REASON_CHARGER_SHUTDOWN, so this code:

```
         if (dinDocDec.V2G_Message.Body.CurrentDemandRes.DC_EVSEStatus.EVSEStatusCode == dinDC_EVSEStatusCodeType_EVSE_Shutdown)
         {
            /* https://github.com/uhi22/pyPLC#example-flow, checkpoint 790: If the user stops the
               charging session on the charger, we get a CurrentDemandResponse with
               DC_EVSEStatus.EVSEStatusCode = 2 "EVSE_Shutdown" (observed on Compleo. To be tested
               on other chargers. */
            addToTrace(MOD_PEV, "Checkpoint790: Charging is terminated from charger side.");
            setCheckpoint(790);
            pev_isUserStopRequestOnChargerSide = 1;
            Param::SetInt(Param::StopReason, STOP_REASON_CHARGER_SHUTDOWN);
         }
```

Most interesting part of log:

```
23:30:09.049 [cha] 102.Status changed 0x89 -> 0x81
23:30:09.050 [cha] Adapter contactor closing
23:30:09.053 [cha] ====>>>> set state WaitForCarAskingAmps
23:30:09.056 [cha] state:WaitForCarAskingAmps cycles:0 charger: out:345V/0A avail:461V/200A/200A rem_t:15300s thres=435V st=0x24 car: ask:0A cap=34.50kWh est_t:1m err:0x0 max:435V max_t:15300s min:6A soc:19% st:0x81 pn:2 target:410V batt:346V
23:30:09.075 [cha] 109.OutputVoltage changed 0 -> 345
23:30:09.235 [26114] PreCharge aknowledge received.
23:30:09.237 PreCharge aknowledge received. Inlet 345V, accu 346V, uMin 0V
23:30:09.242 [26123] Difference between accu voltage and inlet voltage is small.
23:30:09.247 hardwareInterface_setPowerRelayOn
23:30:09.534 [26414] [CONNMGR] 165 0 0 36 0 150 320 --> 100
23:30:09.715 [26594] Contactors assumingly finished closing. Sending PowerDeliveryReq.
23:30:09.775 [26655] Checkpoint700: Starting the charging loop with CurrentDemandReq
23:30:09.849 [cha] 109.OutputCurrent changed 0 -> 1
23:30:09.864 [26745] Checkpoint790: Charging is terminated from charger side.
23:30:09.870 [26751] User requested stop on charger side. Sending PowerDeliveryReq Stop.
23:30:09.926 [26805] Giving the charger some time to detect StateB and ramp down the current.
23:30:09.948 [cha] 109.OutputCurrent changed 1 -> 0
23:30:09.951 Ccs stop reason 0x4. Power off pending...
23:30:10.049 [cha] Stopping before starting
```

Car is waiting/thinking before it will ask for amps. Car has supposedly closed its contactors, at least it said so via flag CONTACTOR_OPEN_OR_WELDING_DETECTION_DONE = false, but it may be lying... Maybe they are not really closed before asking for amps.

The charger enter PowerDelivery and right away try to deliver current (at least it set Param::EvseCurrent to 1), seems to violate spec. It then immediately fail, for reasons unknown, fail. Could it be because it detect an open circuit? (if car has not physically closed the contactors yet?).

Where to go from here:
Not sure. This charger is a weird nut. But an experiment, where ccs CurrentDemand is not started until chademo chargingLoop is entered, could be interesting to try. Then the "charger detect an open circuit" theory should be proved or not.

dalathegreat, 2025-07-23:
@Kazhuu , this charger in Finland is indeed notoriously unreliable. The brand is "Plugit HUBE S". It is one of the few chargers that no version of the Longgood firmware worked with, and as a kicker it also fails quite a bit with native CCS cars. So @osexpert is correct, the firmware on the charger should definately be improved first.

uhi22, 2025-07-23:
There are chargers around (eg ABB) which are not able to regulate the voltage at zero load. We have seen voltage overshoots, and the charger shutted down due to overvoltage detection. In normal case this should not happen because the contactors shall be closed BEFORE sending the first CurrentDemand.
It would be interesting to log the EVSEPresentVoltage.
From my pov the EVSEPresentCurrent of 1A is just a random value because 1A is nearly nothing compared to 200 or more amps full range.

Kazhuu, 2025-07-23:
> [@Kazhuu](https://github.com/Kazhuu) , this charger in Finland is indeed notoriously unreliable. The brand is "Plugit HUBE S". It is one of the few chargers that no version of the Longgood firmware worked with, and as a kicker it also fails quite a bit with native CCS cars. So [@osexpert](https://github.com/osexpert) is correct, the firmware on the charger should definately be improved first.
> 
> (If possible, rename the title of this issue to "Failed to charge on Plugit HUBE S")

Thanks @dalathegreat for confirming this. I was there almost at midnight so maybe I missed the plate displaying the brand. I changed the tittle of the issue.

> Trial and error: trying to ask for minimum 1A and see if it makes any difference:
[osexpert/ccs32clara-chademo/actions/runs/16468826663](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16468826663)

I can give this a try to see does it make any difference. I could also give the longgood firmware a try on this one to see does it work.

osexpert, 2025-07-23:
> There are chargers around (eg ABB) which are not able to regulate the voltage at zero load. We have seen voltage overshoots, and the charger shutted down due to overvoltage detection. In normal case [...]

I think it has to do with the timing than 1A in itself. When I look at some of the working logs I have, Checkpoint700: Starting the charging loop with CurrentDemandReq is after the chademo charging loop has started, so asking amps will always be > 0 in this case. But this charger, it seems to be "insanely" fast, so I see Checkpoint700: Starting the charging loop with CurrentDemandReq "long" before chademo charging loop has started, so default asking amps would then be 0. In itself, not a problem, but I am guessing some chargers may (incorrectly IMO) be using 0 asking amps as a trigger to stop charging. So by using 1A as default, this will be the value used before the chademo charging loop (eventually) start.

But its all speculations, it will be interesting to see the result:-)

Kazhuu, 2025-07-24:
Thanks to look into this @osexpert. I could still try this again if  you have an idea what to try. I could also try to poke the code myself and compile it and see does it work, but I'm not familiar with the Chademo or CCS standards at all. Hence I can try but it will take longer time.

uhi22, 2025-07-24:
I'm clearly a fan of measuring before patching. Do we have a reason to not log the raw ethernet or tcp traffic?

Other topic regarding the code above: DC_EVSEStatus_isUsed indicates that the DC_EVSEStatus is contained in the message. It leads to unpredictable results, if we evaluate DC_EVSEStatus while DC_EVSEStatus_isUsed is False.

Kazhuu, 2025-07-24:
Can you make a commit of the code @osexpert so I get the binary from the CI to try this out.

Kazhuu, 2025-07-24:
I tried the binary and did not work. Here is the log if it's any help.

[serial_20250724_233837.txt](https://github.com/user-attachments/files/21419048/serial_20250724_233837.txt)

osexpert, 2025-07-24:
Thank you. Then two of my theories are ruled out and I have no one left:-) So probably the next step is to turn on extensive logging, possibly if exi (binary xml format), and see if this can help us somehow.

Kazhuu, 2025-07-25:
Is this logging something you do with this binary and I could try again?

osexpert, 2025-07-25:
Looking at the original FW, the timing matches pretty much with the custom FW:
```
22:38:49.180 PowerDelivery ok
22:38:49.313 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.393 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.438 CHAdeMO max_v:435 b_v:410 r_i:0 SOC:63 max_t:175 SF_f:24 sta:81
22:38:49.473 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.553 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.637 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.717 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.797 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.878 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:49.948 CHAdeMO max_v:435 b_v:410 r_i:2 SOC:63 max_t:175 SF_f:24 sta:81
22:38:49.960 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:50.041 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:0 cumuT:0 
22:38:50.133 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:175 cumuT:0 
22:38:50.213 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:175 cumuT:0 
22:38:50.293 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:175 cumuT:0 
22:38:50.374 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:0 remaT:175 cumuT:0 
22:38:50.457 CurrentDmax_v:435 b_v:410 r_i:12 SOC:63 max_t:175 SF_f:5 sta:81
22:38:50.546 CurrentDemand MaxV:500 MaxI:200 EVSE:410 I:4 remaT:175 cumuT:0 
```
Here also CurrentDemand with 0A start before car start asking for amps (I:4). And the timing should be ok, and asking for 0A should be ok and we know PowerDelivery is ok. So then logging is the next step. Possibly I will make a branch with exi logging enabled. Thanks.

osexpert, 2025-07-25:
Trying to see what the spec says about CurrentDemand.

@uhi22 do you know the history behind setting these in CurrentDemand?        
BulkChargingComplete\RemainingTimeToFullSoC\RemainingTimeToBulkSoC

pyplc does not set it:
    def sendCurrentDemandReq(self):
        soc = str(self.hardwareInterface.getSoc())
        EVTargetCurrent = str(self.hardwareInterface.getAccuMaxCurrent())
        EVTargetVoltage = str(self.hardwareInterface.getAccuMaxVoltage())
        msg = addV2GTPHeader(self.exiEncode("EDI_"+self.sessionId + "_" + soc + "_" + EVTargetCurrent + "_" + EVTargetVoltage )) # EDI for Encode, Din, CurrentDemandReq
        self.addToTrace("responding " + prettyHexMessage(msg))
        self.Tcp.transmit(msg)
        
Also example raw log does not set it in pyplc: \pyPLC\RawExiLog.txt.decoded.txt

It is probably not harmful, just curious.

Also tried to ask chatgpt: 

`some DIN 70121-based chargers may fail or behave unexpectedly if optional fields like RemainingTimeToBulkSoC or RemainingTimeToFullSoC are included`

```
Known Scenarios
    Some early implementations of DIN 70121 chargers in Europe have been known to fail without EVMaximumCurrentLimit.
    Some EVSEs use EVMaximumVoltageLimit to validate the EVTargetVoltage, and reject the message if no upper limit is provided, especially if the requested voltage is high (e.g., >400 V
```

It also think that there should be at least 100ms between every CurrentDemandReq sent.

Probably not the most reliable source. And its all just speculations:-)

Kazhuu, 2025-07-25:
> Added exi logging here: [osexpert/ccs32clara-chademo/actions/runs/16518422859](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16518422859) Untested, as usual:-)

Here is the log from this version of the firmware. Failed to charge as usual. 

[serial_20250725_235702.txt](https://github.com/user-attachments/files/21437674/serial_20250725_235702.txt)

I think I clearly here that the contactors of the adapter get closed and then it seems the charge is about to start. Our Nissan Leaf is flashing the LEDs on the dashboard about the current capacity of the charge. But then very soon the charging will fail and soon after I hear the contactors open. Not sure is the any helpful.

osexpert, 2025-07-25:
Thanks. This is getting stranger and stranger. In this case, it did not even reach CurrentDemand:

23:56:32.414 [31258] Contactors assumingly finished closing. Sending PowerDeliveryReq.
23:56:32.418 encoded exi has 20 bytes: 80 9A 02 00 00 00 00 09 2F 85 A8 51 60 00 40 80 30 01 82 87
23:56:32.444 decoder will see has 18 bytes: 80 9A 02 00 00 00 00 09 2F 85 A8 51 40 04 20 80 18 40
23:56:32.448 PowerDeliveryResponse ResponseCode:0 DC_EVSEStatus_isUsed:1 EVSEStatusCode:2

2: dinDC_EVSEStatusCodeType_EVSE_Shutdown

Because of my previous experiment in PowerDelivery, it is now going in loop here, until I assume it goes into timeout.
So now it is failing even earlier than before, just because I enabled logging. It make no sense at all :-) Logging should add delay and then I would expect things to fail later rather than sooner :-D

I think we need some time to chew on this, until we decide the next step.

Kazhuu, 2025-07-26:
Interesting. If you come up with something then I can go test again. So there is no other way to figure this out other than a bit of trial and error I suppose. I should probably dig into the standards as well to understand more of this what is going on.

osexpert, 2025-07-26:
I used https://github.com/uhi22/OpenV2Gx (I see there is a PR here?)
and sligthly modified claralogConverter.py

Decoded here:

[serial_20250725_235702.claralog.decoded.txt](https://github.com/user-attachments/files/21446504/serial_20250725_235702.claralog.decoded.txt)

Strange things:

One message fails to decode?
OpenV2G.exe DD809A0200000000092F85A85080000000
"msgName": "ChargeParameterDiscoveryRes",
"info": "16 bytes to convert",
"error": "runTheDecoder error-100",

All bytes say they are res? Where are the req? I am confused.

```
23:56:32.418 encoded exi has 20 bytes: 80 9A 02 00 00 00 00 09 2F 85 A8 51 60 00 40 80 30 01 82 87
{
"msgName": "PreChargeRes",
"info": "20 bytes to convert", 
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "0000000024be16a1",
"header.Notification_isUsed": "0",
"header.Signature_isUsed": "0",
"ResponseCode": "OK",
"DC_EVSEStatus.EVSEIsolationStatus": "1",
"DC_EVSEStatus.EVSEIsolationStatus_isUsed": "1",
"DC_EVSEStatus.EVSEStatusCode": "1",
"EVSEStatusCode_text": "EVSE_Ready",
"DC_EVSEStatus.NotificationMaxDelay": "3",
"DC_EVSEStatus.EVSENotification": "0",
"EVSEPresentVoltage.Multiplier": "0",
"EVSEPresentVoltage.Value": "96",
"EVSEPresentVoltage.Unit": "V",
"debug": "Line490"
}
```

Where do suddenly 96 volts come from?

Kazhuu, 2025-07-28:
> I used an incorrect variable when logging "encoded exi", so they were wrong:-D Fixed now.
> 
> All "decoder will see" should be fine thou.
> 
> [osexpert/ccs32clara-chademo/actions/runs/16543394585](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16543394585)
> 
> Took a quick look at the "decoder will see" messages, and they look fine to me... And then...boom: charger shutdown.

Hello! I'll try this later today. I was bit in the hurry to not have time to test this. I'll put a log here once I get a change to go there your changes.

osexpert, 2025-07-28:
> Using Chademos target voltage as "normal battery full voltage" may be wrong if the car would ramping up the target voltage.

To me it seems from log of original fw always precharge to target voltage (410) so it probably wont hurt as an experiment to try precharge to target - 10 (400), But I agree it is not right, so if this is what it takes to make it work...its still not a good solution.

osexpert, 2025-07-29:
As I mentioned before:

I wondered what was the point of this in pev_sendCurrentDemandReq:
```
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.BulkChargingComplete_isUsed = 1u;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.BulkChargingComplete = 0u; /* not complete */
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToFullSoC_isUsed = 1u;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToFullSoC.Multiplier = 0;  /* -3 to 3. The exponent for base of 10. */
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToFullSoC.Unit = dinunitSymbolType_s;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToFullSoC.Unit_isUsed = 1;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToFullSoC.Value = 1200; /* seconds */

   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToBulkSoC_isUsed = 1u;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToBulkSoC.Multiplier = 0;  /* -3 to 3. The exponent for base of 10. */
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToBulkSoC.Unit = dinunitSymbolType_s;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToBulkSoC.Unit_isUsed = 1;
   dinDocEnc.V2G_Message.Body.CurrentDemandReq.RemainingTimeToBulkSoC.Value = 600; /* seconds */
```

This info is also completely invisible in the decoded information... It seems to be included in the binary data, at least compared to 2022-12-20_alpi_currentDemandSequenceErrorShutdown.decoded.txt that encode a CurrentDemandReq with 23 bytes, this log encode CurrentDemandReq into 35 bytes, so it has extra data that the decoder seem to not show...
So possibly the decoder is not decoding all the information? It is weird.

Kazhuu, 2025-07-29:
Nice update! Do you have some update again I could test?

osexpert, 2025-07-29:
I used a different decoder (a random one, https://github.com/FlUxIuS/V2GDecoder), and it show the complete message:
```
<?xml version="1.0" encoding="UTF-8"?>
<ns7:V2G_Message xmlns:ns7="urn:iso:15118:2:2010:MsgDef" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:ns3="http://www.w3.org/2001/XMLSchema" xmlns:ns4="http://www.w3.org/2000/09/xmldsig#" xmlns:ns5="urn:iso:15118:2:2010:MsgBody" xmlns:ns6="urn:iso:15118:2:2010:MsgDataTypes" xmlns:ns8="urn:iso:15118:2:2010:MsgHeader">
	<ns7:Header>
		<ns8:SessionID>000000007EE3873B</ns8:SessionID>
	</ns7:Header>
	<ns7:Body>
		<ns5:CurrentDemandReq>
			<ns5:DC_EVStatus>
				<ns6:EVReady>true</ns6:EVReady>
				<ns6:EVErrorCode>NO_ERROR</ns6:EVErrorCode>
				<ns6:EVRESSSOC>22</ns6:EVRESSSOC>
			</ns5:DC_EVStatus>
			<ns5:EVTargetCurrent>
				<ns6:Multiplier>0</ns6:Multiplier>
				<ns6:Unit>A</ns6:Unit>
				<ns6:Value>12</ns6:Value>
			</ns5:EVTargetCurrent>
			<ns5:BulkChargingComplete>false</ns5:BulkChargingComplete>
			<ns5:ChargingComplete>false</ns5:ChargingComplete>
			<ns5:RemainingTimeToFullSoC>
				<ns6:Multiplier>0</ns6:Multiplier>
				<ns6:Unit>s</ns6:Unit>
				<ns6:Value>1200</ns6:Value>
			</ns5:RemainingTimeToFullSoC>
			<ns5:RemainingTimeToBulkSoC>
				<ns6:Multiplier>0</ns6:Multiplier>
				<ns6:Unit>s</ns6:Unit>
				<ns6:Value>600</ns6:Value>
			</ns5:RemainingTimeToBulkSoC>
			<ns5:EVTargetVoltage>
				<ns6:Multiplier>0</ns6:Multiplier>
				<ns6:Unit>V</ns6:Unit>
				<ns6:Value>410</ns6:Value>
			</ns5:EVTargetVoltage>
		</ns5:CurrentDemandReq>
	</ns7:Body>
</ns7:V2G_Message>
```
So OpenV2G.exe seem to be broken...
OpenV2G.exe DD809A02000000001FB8E1CED0D1400580C0C0306001810580480C08360100C142680C00
```
{
"msgName": "CurrentDemandReq",
"info": "35 bytes to convert",
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "000000007ee3873b",
"header.Notification_isUsed": "0",
"header.Signature_isUsed": "0",
"DC_EVStatus.EVReady": "1",
"DC_EVStatus.EVErrorCode": "0",
"DC_EVErrorCodeText": "NO_ERROR",
"DC_EVStatus.EVRESSSOC": "22",
"EVTargetCurrent.Multiplier": "0",
"EVTargetCurrent.Value": "12",
"EVTargetCurrent.Unit": "A",
"EVMaximumVoltageLimit_isUsed": "0",
"EVMaximumCurrentLimit_isUsed": "0",
"EVMaximumPowerLimit_isUsed": "0",
"ChargingComplete": "0",
"EVTargetVoltage.Multiplier": "0",
"EVTargetVoltage.Value": "410",
"EVTargetVoltage.Unit": "V",
"debug": "Line9057Line9260Line436"
}
```

"Do you have some idea or update again I could test? Would it possible to try how the original firmware does it even that is not correct? If I understood your comments correctly :D Just so see this works?"
Yes...I have avoided increasing precharge voltage, even if the original fw does this, and tried many other things. But it seems the time has come to try it, since (almost) all other ideas have been tested:-/

Kazhuu, 2025-07-30:
> Yes...I have avoided increasing precharge voltage, even if the original fw does this, and tried many other things. But it seems the time has come to try it, since (almost) all other ideas have been tested:-/

If you provide me with the changes I can go try. If that does work, then we are wiser what is the issue.

Kazhuu, 2025-07-31:
Hello I did couple some tests yesterday with your changes and on my own. I tried to apply the changes you suggested on top of the `debug/exi_logging` branch. But it seems all three options above causes some infinite loop in the logs and the charging station is talking with the adapter for a long time until I stopped it. Here is the log file with one of the `TargetBatteryVoltage` usage, don't remember which one because both them seemed to cause  the same thing.

[test2-and-test3-on-lexi-branch.txt](https://github.com/user-attachments/files/21522781/test2andtest-on-lexi-branch.txt)

Then I tried to apply the "worse" case on top of the main branch but I think it produced the similar result.

[test3-on-main.txt](https://github.com/user-attachments/files/21522802/test3-on-main.txt)

Any ideas?

Kazhuu, 2025-08-02:
Nice! When you have time to do that? I'll leave away for a week tomorrow so I'll not be able to test things for that time.

osexpert, 2025-08-02:
Hi, the not-working-charging.txt is the same as I saw before, that there is no contact with the charger (same as like charger not plugged into adapter). So this is a completely unrelated problem that is just interfering with the testing. This problem may be hardware related and not possible to fix.

osexpert, 2025-08-02:
In rare cases, it may be necesary to start the adapter first, and then plug the charger into the adapter. Or plug the charger out and then in again. You can get a clue about this from the led blinking on the adapter. When contact with charger is made, one short blink. So if adapter never get past the initial slow blinking, there is no contact with charger (analog to charger not being plugged in).

This may be related with quirky proximity detection, and is hardware related.

Personally, I never had this problem, but with this type of charger, I guess anything can happen, as it is already breaking the standard in one way.

osexpert, 2025-08-02:
not-working-charging.txt
```
14:50:44.459 [18097] [SDP] initiating SDP request
14:50:44.700 [18337] [SDP] ERROR: Did not receive SDP response. Giving up.
14:50:44.729 [18367] [PEVSLAC] Checking whether the pairing worked, by GET_KEY.REQ...
14:50:44.734 [18373] [PEVSLAC] from 12 entering 13
14:50:44.737 [18377] [SDP] Checkpoint200: Starting SDP.
14:50:44.761 [18398] [SDP] initiating SDP request
14:50:44.968 [18606] [CONNMGR] 165 0 0 234 0 0 0 --> 15
14:50:45.028 [18666] [PEVSLAC] It was sufficient time to get the answers from the modems.
14:50:45.034 [18672] [PEVSLAC] No EVSE seen (yet). Still waiting for it.
14:50:45.039 [18678] [PEVSLAC] from 13 entering 12
```
So I was wrong, it has contact with charger, but it can't create a "good" connection. This is some very low level communication problem, so it is still unrelated to any of the changes I have (ever) done:-)

Kazhuu, 2025-08-02:
Okay the communication error might be something else than the commit on the main I said earlier. As I tested the latest main and it included the commit I was talking about and the communication with the charger worked fine. I tested the main on this commit https://github.com/osexpert/ccs32clara-chademo/commit/c7f98d84b30370c099240c67fb4602627ca2685a. As usual, stopping the charging from the station did not work without it reporting a fault.

Here is the log from the working charge session. This does not include the end communication as I stopped the logging before that.

[working-main.txt](https://github.com/user-attachments/files/21561517/working-main.txt)

Do you have some idea what could be wrong with the communication when stopping charging from the station itself?

osexpert, 2025-08-03:
New experiment for stop charging error (try to implement tcp fin):
https://github.com/osexpert/ccs32clara-chademo/actions/runs/16704056425

osexpert, 2025-08-04:
Updated version:-) https://github.com/osexpert/ccs32clara-chademo/actions/runs/16721219895
Not tested at all, but I will try to test it today so you don't have to test something that is obviously not working:-) Thanks.

Kazhuu, 2025-08-10:
Okay finally I was able to test this as I'm back home. With this change https://github.com/osexpert/ccs32clara-chademo/actions/runs/16775904330 the charging succeeded, but the ending the charging from the station were reported as faulted in the charging station screen. Here is the log of the charging session. Any idea for the next steps?

[charge-succeeded-ending-failed.txt](https://github.com/user-attachments/files/21707344/charge-succeeded-ending-failed.txt)

osexpert, 2025-08-10:
Hi. The log logs pretty good. 3x FIN in reply may be a bit strange,I will look into that. Chatgpt suggest a delay between recieve stopsessionrequest and tcp_close, may be worth a try.

What does the error say exactly? Do you still meed to pay for the cjarging? Can it be a free charging loophole/bug in the charger?

I still want to get this changes into main as FIN is nice to have. I will then make a new main branch for experiments. I will probably have a new experiment for you in a couple of days :-)

Kazhuu, 2025-08-11:
> What does the error say exactly? Do you still need to pay for the charging? Can it be a free charging loophole bug in the charger?

The charged is just flashing blue light and the screen says "faulted". After little bit of flashing it just returns to the normal state that it was before. 

The free charging is actually a good idea to test. All the tests have been costing me 0€ because I have been charging a very short period of time I think. I could try again just out of curiosity to charge longer and then stop the charge so that it faults the communication. I'll post the result here once tested.

> I still want to get this changes into main as FIN is nice to have. I will then make a new main branch for experiments. I will probably have a new experiment for you in a couple of days :-)

Yes please let's fix this. I'll go test again after you have changes available.

osexpert, 2025-08-11:
I changed my mind and will continue on the same branch:-)
New experiment: 20250811-99 https://github.com/osexpert/ccs32clara-chademo/actions/runs/16877150424
Do not RST or FIN the connection at all, just ghost it... (this is what ccs32clara originally did, but I wanted to improve it...)
Use a random port instead of a fixed port.

It may be that the charger error is not related to tcp at all...but rather some other thing that happened earlier. Hard to say what...

osexpert, 2025-08-11:
Idea for next experiment:
In: WeldingDetectionReq, EVReady = 1

But from chatgpt (EVReady must be 0 in WeldingDetectionReq) and logs from pyPLC (polestar2), 1 seems wrong here. Most/all changers does not care, but worth an experiment, unless 20250811-99 works.

Kazhuu, 2025-08-13:
> I changed my mind and will continue on the same branch:-) New experiment: 20250811-99 [osexpert/ccs32clara-chademo/actions/runs/16877150424](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16877150424) Do not RST or FIN the connection at all, just ghost it... (this is what ccs32clara originally did, but I wanted to improve it...) Use a random port instead of a fixed port.
> 
> It may be that the charger error is not related to tcp at all...but rather some other thing that happened earlier. Hard to say what...

Okay I tested this revision and the charge still failed. Here is the log of it

[serial_20250813_222511.txt](https://github.com/user-attachments/files/21761502/serial_20250813_222511.txt)

> I added EVReady = 0 in WeldingDetectionReq in main, so you could also try
[osexpert/ccs32clara-chademo/actions/runs/16931569340](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16931569340)
20250813-105

This one I have not tested yet.

Other important also is that indeed when the end of charge communication fails with this station. They indeed do not charge anything from you!!! Which is quite serious loop hole. I did three tests. One I stopped from the app and the station, both ended up in the faulted state and the app reported 0€ charge from me. Then in the last step I stopped the charge from the stop button on the adapter and the charge ended successfully and the app reported some charged euros. I did about 2kW charge on each test to test this. I assume most of the people would stop the charging from the app and then in that case the communication would fail and they are not charged at all for the energy. 

This make me wonder how many other brands have this similar issue if you write a adapter software that intentionally fails the communication at the end...

Kazhuu, 2025-08-13:
> I added EVReady = 0 in WeldingDetectionReq in main, so you could also try [osexpert/ccs32clara-chademo/actions/runs/16931569340](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16931569340) 20250813-105 If both 20250811-99 and 20250813-105 give error at stop charging....idea for next experiment is improved tcp that should handle all FIN states correctly. Thanks.

Okay here is the log for this one as well. The charging failed as usual.

[serial_20250813_231927.txt](https://github.com/user-attachments/files/21761800/serial_20250813_231927.txt)

Kazhuu, 2025-08-16:
> I tested [osexpert/ccs32clara-chademo/actions/runs/16944962782](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16944962782) 20250813-108 from experiments/tcp_fin_proper and it charged on a HYC150 without issues, so it should be ok to try.

I tested this build and it failed communicating at the end. Here is the log

[first.txt](https://github.com/user-attachments/files/21817398/first.txt)

> Also I updated debug/tcp_fin_experiment to not even reply to FIN's (ghost harder, very similar to how clara worked originally):
[osexpert/ccs32clara-chademo/actions/runs/16966363796](https://github.com/osexpert/ccs32clara-chademo/actions/runs/16966363796) 20250814-110

Then I tested this one and it also failed as before. Here is the log.

[second.txt](https://github.com/user-attachments/files/21817403/second.txt)

Any further ideas?

osexpert, 2025-08-16:
I need to think about the next move, if any. But the original fw do not give error after charging?

Kazhuu, 2025-08-16:
I could not get the original firmware even to start the charge now for some reason. After couple of tried I gave up. I can try again later with the original one. I cannot remember what the original firmware even did to be honest. Would need further testing for sure. 

>  Does it matter if stopped via charger, app or adapter stop button?

With this firmware the communication fails at the end from the charger and the app, but not from the adapter stop button.

I also tried to charge longer this time like 20 mins to then faulted the communication at the end. They actually charge you normally. So I think their software is sound, but for some reason they don't charge you when the charge for like 2kW only and then fail the communication. When the stop button from the adapter is used for 2 KW, they will charge you normally. At least that what the app reports.

Kazhuu, 2025-08-19:
Yeah that is true and the most important thing is indeed that the charging works. Any ideas of any further things to test with this? And hopefully someone from the company would contact us regarding this, what could be the potential issue...

osexpert, 2025-08-19:
I checked again, the logs exi's and timings are spotless. And the tcp disconnect experiments did not make a difference. Its hard to fix a problem that is invisible. So I am out of idea.

Kazhuu, 2025-08-19:
Then let's hope the company will response something. Or if not then I'll spam them again.

osexpert, 2025-09-04:
Hi, I am doing cleanup and will close this.
If you want, you can eg. test the latest version
20250903-137
https://github.com/osexpert/ccs32clara-chademo/actions/runs/17444003632
and see if the problem persist, and in case open a new issue (eg. "Error after successful charging on Plugin HUBE S"), and include a picture of the charging station error/screen, if possible.

closed 2025-09-04 by osexpert

osexpert, 2025-09-09:
The log looks a bit different, a tcp fin from charger is not logged. But if it did not complain, I am happy:-)

Nrw release, not sure when. Still a couple of open issues I want to close.

Thanks.