osexpert/ccs32clara-chademo #12: Gave an error code after charging at Alpitronic HY150

conradkwu, 2025-07-29:
[stock charger - working (got code afterwards).txt](https://github.com/user-attachments/files/21489569/stock.charger.-.working.got.code.afterwards.txt)
[jamies faulty before clearing code.txt](https://github.com/user-attachments/files/21489570/jamies.faulty.before.clearing.code.txt)

[image]

Any insights appreciated.

conradkwu, 2025-08-02:
> I added some changes here 2159903
> that should make adapter turn of faster in case of errors/timeout before
> charging has started. Purely cosmetic, but waiting is boring:-)

Will give it a try, thanks!

conradkwu, 2025-08-03:
Same model but different site, it refused to charge at all. 

[serial_20250803_112525.txt](https://github.com/user-attachments/files/21565422/serial_20250803_112525.txt)

osexpert, 2025-08-03:
I am not sure if I ever charged at a Alpitronic HY150. Do you think that Alpitronic HY150 in general (or always) does not work?

osexpert, 2025-08-03:
A working log from original fw could be helpful too. I will see if there is a HYC150 nearby I can test on, else I need help with testing.

Failing points:
jamies faulty before clearing code.txt
11:47:09.277 [13758] Checkpoint530: Will send ContractAuthenticationReq
11:47:09.514 [13993] [TCP] FIN received, sending final ACK

serial_20250803_112525.txt:
11:25:06.879 [85651] Checkpoint700: Starting the charging loop with CurrentDemandReq
11:25:06.908 [85680] [TCP] RST received, closing connection

leap24 communication failed.txt
09:20:10.201 [36860] Checkpoint700: Starting the charging loop with CurrentDemandReq
09:20:10.231 [36889] [TCP] RST received, closing connection

So basically...the charger keeps closing the tcp connection. Thats pretty rude:-)

osexpert, 2025-08-03:
I made a speculative change/experiment (untested) in 20250803-84 https://github.com/osexpert/ccs32clara-chademo/actions/runs/16708255138

It will now respond to keep-alive probes sent from the changer. Just a wild guess that this may be the problem.

Thanks.

conradkwu, 2025-08-08:
Havent flashed to the latest version yet but here is stock B49.4 for your reference. Cheers. 

[alpitronic 150 b49.4 ok.txt](https://github.com/user-attachments/files/21688206/alpitronic.150.b49.4.ok.txt)

osexpert, 2025-08-10:
Too bad....its the same problem as before:
```
14:42:44.702 [57449] Contactors assumingly finished closing. Sending PowerDeliveryReq.
14:42:45.122 [57869] [CONNMGR] 165 0 0 0 0 120 300 --> 100
14:42:45.287 [cha] state:WaitForCarAskingAmps cycles:11 charger: out:358V/0A avail:461V/200A/200A rem_t:15300s thres=435V st=0x24 car: ask:0A cap=53.20kWh est_t:1m err:0x0 max:435V max_t:15300s min:6A soc:80% st:0x81 pn:2 target:410V batt:359V
14:42:45.753 [58500] Checkpoint700: Starting the charging loop with CurrentDemandReq
14:42:45.782 [58530] [TCP] RST received, closing connection
```
I had a feeling this was a tcp problem, since tcp is being reset, but it may also be unrelated to the tcp...maybe the charger just use a very "unusual" (non standard and spec violating) way to stop the charging/signal a problem.

Next time I suggest to use
20250806-91
https://github.com/osexpert/ccs32clara-chademo/actions/runs/16787187672
that has raw data (exi) logging enabled and this info may help to figure out what is going on here...

Thanks.

osexpert, 2025-08-10:
Hi,

There seems to be 2 (or 3?) problems here now, one is an error et the end of the charging?
"1. It works on the first try, then Session Aborted shown on charging stall when finished"
This sounds similar to what happens in issue 11 [Failed to charge on Plugit HUBE S](https://github.com/osexpert/ccs32clara-chademo/issues/11)
There is a test version available for that issue, that uses tcp FIN instead of tcp RST to end tcp: 20250806-90 https://github.com/osexpert/ccs32clara-chademo/actions/runs/16775904330

And the second problem, seem to be with repeated charging, where it fail on second try? It may also be related to tcp RST vs FIN, so try 20250806-90 for this as well.

The third was already mentioned, and AFAICS its not related to repeated charging: https://github.com/osexpert/ccs32clara-chademo/issues/12#issuecomment-3172691242 and here using 20250806-91 will give extra logging (but note: 20250806-91 does not have the tcp FIN experiment)

Thanks.

conradkwu, 2025-08-10:
I first tried 0806-90 and session aborted.

Then I flashed to -91 and it worked! However for some reason the serial-USB terminal keeps quitting so I could not capture the handshake the first 2 times.

On the third time it fails to work again.

And now the adaptor wont turn off and keep on looping.

Thanks for your effort.  

[hyc 150 90 fail.txt](https://github.com/user-attachments/files/21707120/hyc.150.90.fail.txt)
[hyc 150 91 success.txt](https://github.com/user-attachments/files/21707121/hyc.150.91.success.txt)
[hyc 150 91 success 2nd attempt.txt](https://github.com/user-attachments/files/21707122/hyc.150.91.success.2nd.attempt.txt)
[hyc 150 91 fail.txt](https://github.com/user-attachments/files/21707123/hyc.150.91.fail.txt)
[hyc 150 91 fail 2nd attempt.txt](https://github.com/user-attachments/files/21707124/hyc.150.91.fail.2nd.attempt.txt)
[infinite loop 91 after charging at hyc 150.txt](https://github.com/user-attachments/files/21707125/infinite.loop.91.after.charging.at.hyc.150.txt)

osexpert, 2025-08-11:
Very nice:-)

Since repeated changing failed, it seemed the charger was somehow incorrectly remembering something about us, from the last charging session... So it sees us again, and start to mess with us, sending us random resets and stuff, not very nice...

I assume it still does these things, but it will now send it to the port we used last time, and since using a new random port, it wont hurt us.

The chances of choosing the same random port 2 times in a row exist, but is slim.

conradkwu, 2025-08-11:
I left the error codes from 90/91 uncleared on purpose, they are still there while using 99 but it just works.

I will try another HYC charger later. 

[image]

conradkwu, 2025-08-11:
It still gives error codes after I cleared it on LeafSpy, but nonetheless it works flawlessly in another HYC150 and HYC300.

Just throw in another ABB for reference.

Is this good to go or you will make a release version of it? cheers. 

[image]

[lidl abb 99 success.txt](https://github.com/user-attachments/files/21718754/lidl.abb.99.success.txt)
[shell hyc 300 99.txt](https://github.com/user-attachments/files/21718755/shell.hyc.300.99.txt)
[leap24 hyc150 99.txt](https://github.com/user-attachments/files/21718756/leap24.hyc150.99.txt)

osexpert, 2025-08-11:
Hi,

"It still gives error codes after I cleared it on LeafSpy" Hmm...that's not good. Which of the attached logs made the error appear?
But with -36 it wont charge at all, so its _kind of_ an improvement, even if it give an error in the car...but it would be best to figure it out.

I had planned a new release when/if issue 11 [Plugit HUBE S](https://github.com/osexpert/ccs32clara-chademo/issues/11) is resolved. Since -99 is also from the test branch used for issue 11 experiments, it wont be released (directly), but eventually the good parts of this branch will end up in main and there will be a release (candidate) :-)

conradkwu, 2025-08-11:
Kudos to you!

I cleared codes before shell hyc300 then the new + old ones popped up as per screenshot.

conradkwu, 2025-08-12:
Happened to be at a HYC_50 and it was good. 

[mfg hyc50 -99 success.txt](https://github.com/user-attachments/files/21740395/mfg.hyc50.-99.success.txt)

osexpert, 2025-09-04:
This is most likely fixed now, so closing it. If you test 20250818-112 and still experience the problem, please open a new issue. Thanks.