osexpert/ccs32clara-chademo #17: Hyperchargers: WeldingDetectionRes, EVSEPresentVoltage never drops

osexpert, 2025-08-14:
On Hyperchargers (eg. HYC150, HYC300), WeldingDetectionRes.EVSEPresentVoltage never drop below the last voltage in CurrentDemandReq.
It eventually time out and go into safe shutdown, so it still works, but it should not be like this.
Maybe try to wait even longer for the voltage to drop? Or detect that the voltage is the same as in was last in CurrentDemandReq? (meaning, it is never updated)
Or can it be that the voltage get "stuck" due to something else that happens earlier, that the charger do not like?
Try to dig up existing exi-logs from Hypercharger charging session, preferably with real cars?

eddiew-source, 2025-08-20:
I think I have exactly the same issue 
Firstly thanks for your time and effort on this. It didn’t work with my car at all on the previous firmware and I have had success. 

Sadly I tried a hyper charger this evening and that didn’t work. I have full logs. I shall post them tomorrow

eddiew-source, 2025-08-21:
https://pastecode.io/s/9sf7k08m

heres the logs from mine, chat GPT says its something to do with welding detection
https://www.electromaps.com/en/charging-stations/cyprus/nicosia/strovolos/bmw-hq-nicosia-150kw-meteoron
this is the charger in question

if i can change something internally let me know and ill retry it
thanks

eddiew-source, 2025-08-21:
not a developer at all but ChatGPT has suggested the below changes
diff --git a/src/chademo/pev_states.c b/src/chademo/pev_states.c
index XXXXXXX..YYYYYYY 100644
--- a/src/chademo/pev_states.c
+++ b/src/chademo/pev_states.c
@@
+// -------- Hypercharger-friendly welding detection tuning --------
+// These defaults keep safety conservative while tolerating slow EVSE bus bleed.
+#ifndef WELD_OK_VOLTAGE_V
+#define WELD_OK_VOLTAGE_V       60     // declare OK when EVSEPresentVoltage <= 60 V
+#endif
+#ifndef WELD_MAX_TIME_MS
+#define WELD_MAX_TIME_MS        8000   // wait up to 8 s before giving up waiting
+#endif
+#ifndef WELD_POLL_INTERVAL_MS
+#define WELD_POLL_INTERVAL_MS   80     // poll cadence during WeldingDetection
+#endif
+#ifndef WELD_STALE_WINDOW_MS
+#define WELD_STALE_WINDOW_MS    1200   // if reading hasn't changed for this long, treat as "stale"
+#endif
+#ifndef WELD_STALE_DELTA_V
+#define WELD_STALE_DELTA_V      1      // consider "changed" only if >1 V difference
+#endif
+
@@
-// Existing state handler (simplified):
-// static void stateWaitForWeldingDetectionResponse(...) {
-//   LOG_I("In state WaitForWeldingDetectionRes");
-//   while (true) {
-//     sendWeldingDetectionReq();
-//     int voltage = getEVSEPresentVoltageV();
-//     if (voltage <= threshold) { /* success */ break; }
-//     // retry...
-//   }
-// }
+// Hypercharger-tolerant implementation
+static void stateWaitForWeldingDetectionResponse(/* params... */)
+{
+    LOG_I("In state WaitForWeldingDetectionRes");
+
+    const uint32_t t_start = HAL_GetTick();        // or project tick API
+    uint32_t       t_last_change = t_start;
+    int            last_v = -1;
+
+    for (;;) {
+        // Send/await WeldingDetection response
+        sendWeldingDetectionReq();
+
+        // Sleep/poll according to platform
+        delay_ms(WELD_POLL_INTERVAL_MS);          // or scheduler/postback
+
+        const int v = getEVSEPresentVoltageV();   // must return volts (int)
+        if (v >= 0 && v <= WELD_OK_VOLTAGE_V) {
+            LOG_I("WeldingDetection OK: EVSEPresentVoltage=%dV (<=%dV)", v, WELD_OK_VOLTAGE_V);
+            break; // proceed to SessionStop/Unlock
+        }
+
+        // Track if reading is changing
+        const uint32_t now = HAL_GetTick();
+        if (last_v < 0 || (v >= 0 && ABS(v - last_v) > WELD_STALE_DELTA_V)) {
+            last_v = v;
+            t_last_change = now;
+        } else if ((now - t_last_change) > WELD_STALE_WINDOW_MS) {
+            // Hypercharger quirk: value can sit "stuck" (e.g., 384 V) for a while.
+            LOG_W("WeldingDetection: stale EVSE voltage ~%dV for >%ums; continuing to wait (slow EVSE bleed?)",
+                  v, (unsigned)(now - t_last_change));
+            t_last_change = now; // avoid log spam
+        }
+
+        // Absolute max waiting time (safety: never re-energise; just complete stop)
+        if ((now - t_start) > WELD_MAX_TIME_MS) {
+            LOG_W("WeldingDetection: max wait %ums exceeded; finishing stop anyway (EVSE slow to bleed).",
+                  (unsigned)WELD_MAX_TIME_MS);
+            break; // safe finish; we are powering off / not closing contactors
+        }
+    }
+}
+
+// (Optional) small grace before starting the weld loop after Stop, to let EVSE open contactors:
+// call: delay_ms(300); before entering stateWaitForWeldingDetectionResponse()

uhi22, 2025-09-01:
I checked some logs which I took while charing my Hyundai Ioniq on Alpitronic hyperchargers HYC150 and HYC300 last year (available as pcap at https://github.com/uhi22/Ioniq28Investigations/tree/main/CCM_ChargeControlModule_PLC_CCS). They show, that the EVSEPresentVoltage correctly drops during the WeldingDetection. So I do not see an issue of the hyperchargers.
But maybe they distributed a buggy software in the meanwhile, who knows. Is there a chance to find out the charge point operator of the chargers in question? Maybe pull a log of a charging session with a normal car at this charger?

osexpert, 2025-09-01:
Your logs look fine, its strange for sure. It could be different firmware versions in different countries or different operators thou, maybe you logged the good version:-)

"Maybe pull a log of a charging session with a normal car at this charger?"
Yes, that would be interesting, unfortunately i only have a chademo car.

I did see something interesting in a log thou, I see 
https://github.com/uhi22/Ioniq28Investigations/blob/main/CCM_ChargeControlModule_PLC_CCS/spi_ioniq_alpiHYC150_makerspace_landsberg_right_big_2024-04-03_ok.csv.pcap.decoded.txt
line 52309: 809a02000000001de0c9bb521000000780
This decodes to (at least in my fork https://github.com/osexpert/OpenV2Gx where I added some missing properties):
```
OpenV2G.exe DD809a02000000001de0c9bb521000000780
{
"msgName": "WeldingDetectionReq",
"info": "17 bytes to convert",
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "00000000778326ed",
"DC_EVStatus.EVReady": "0",
"DC_EVStatus.EVErrorCode": "0",
"DC_EVErrorCodeText": "NO_ERROR",
"DC_EVStatus.EVRESSSOC": "60",
"DC_EVStatus.EVCabinConditioning": "0",
"DC_EVStatus.EVRESSConditioning": "0",
"debug": "Line9057Line9090Line9169Line9226Line9260Line556"
}
```
So I will try to add those cabin conditioning properties, and see it makes a difference (it would be really strange if it did thou).
I will see of I can find more differences that could possibly explain it.

Thanks.

osexpert, 2025-09-04:
Adding the cabin conditioning properties did nothing. So when not being able to test/log on real ccs car, I can only assume charger firmware issue.