osexpert/ccs32clara-chademo #20: BMW i5 M60 - Chademo - Issue with Alpitronic hypercharger HYC150

eddiew-source, 2025-08-21:
https://pastecode.io/s/9sf7k08m

heres the logs from mine, 
https://www.electromaps.com/en/charging-stations/cyprus/nicosia/strovolos/bmw-hq-nicosia-150kw-meteoron
https://cdn.shopify.com/b/shopify-inbox/01976983-6906-742b-9333-83020b1d81e5/a0hgitzqs1vpc8k5e670dlq94otu.png?syclid=d2jj3mn10ems73ch4mp0

this is the charger in question, it is almost certaily an Alpitronic hypercharger HYC150

The original firmware of the adapter did not work on ANY charger at all for my car. It just errored and wouldn't even initiate a charge, i have had a successful charge with the car and with this firmware.

However, using this charger at this time, it seemed (sounded, and screen showed) that it was beginning charging, but then stops... i can hear a click in the adapter too when it stops 
thanks

osexpert, 2025-08-21:
here is branch of release 20250813-105 with exi logging enabled:
https://github.com/osexpert/ccs32clara-chademo/actions/runs/17006106136
20250816-111
With such log I can see raw exi data and maybe get a clue what it may be complaining about.

eddiew-source, 2025-08-25:
I have loaded the firmware and should get chance tonight. 

One thing I’ve noticed on the stock firmware I could turn the device on and then in serial output see the firmware version etc 

This doesn’t seem to do that now and just shows me disconnect ? Assume it requires a ccs connection?

eddiew-source, 2025-08-25:
you are indeed correct, i wasn't holding the button for long enough

ccs32clara-chademo 20250816-111 - we are good.. will hopefully get you these tonight - i assume it just pushes the EXI logs via serial? if so i will record my terminal session to make sure i get everything
then i will load the newer firmware you gave and try that

Thanks

osexpert, 2025-08-25:
"i assume it just pushes the EXI logs via serial"
Yes. Good luck:-)

osexpert, 2025-09-04:
There is even a newer version with a lot of experiments here:
20250903-137
https://github.com/osexpert/ccs32clara-chademo/actions/runs/17444003632
Now should automatically correct for deviating voltage.
Thanks.

osexpert, 2025-09-07:
PS: there is no need to test 20250903-137, it is already tested by other and found Non-functional. Thanks.

osexpert, 2025-10-01:
just cleaning up some old issue. All changes/improvements made for this is now also in latest release, so just use latest release. 
If it still not work, just reopen the issue
Thanks.

eddiew-source, 2025-11-22:
Just tried with the latest firmware one the same charger and it doesn’t work I’m afraid. 

It comes up on screen with the vehicle reported an issue. 

I don’t have my laptop so will come back and get the logs.

eddiew-source, 2026-01-06:
> Hi,
>
> A bit of unluck here:
>
> ; ccs cable check fails for unknown reasons. Charger timed out or something. Or something was actually bad with the cable.
> line 385 Will again send CableCheckReq (rc:4 proc:0)
>
> ; Stop button pressed on adapter.
> line 468 Stop button pressed briefly and slac not pending. Power off pending...
>
> ; Ccs was restarted, and trying again, this time the charger return success
> line 867 The EVSE says that the CableCheck is finished and ok.
>
> ; But now it discovers that the stop button had been pressed, and it stops the charging (right before it could have started)
> line 887 Checkpoint700: Starting the charging loop with CurrentDemandReq
> line 888 Power off pending.
>
> line 889 User requested stop on car side. Sending PowerDeliveryReq Stop.
>
> So it seems it the stop button had not been pressed, it could have worked, but we can't know:-)

So I tried it twice here

First time cable error did come up on the unit. So I used the other one. And then it just timed out like it has done before. Stop button hasn’t been pressed

So I think it’s still the charger but I will try again.

eddiew-source, 2026-01-06:
OK so strange update

tried again and it first said error signalised from car

tried again and its now working

annoyingly didnt get the full logs as the baud rate was not set right but i managed to capture after it was charging, though dont know why it failed first time. 

[log.txt](https://github.com/user-attachments/files/24453902/log.txt)

osexpert, 2026-01-21:
There is a newer pre-release in case you want to test:-)
https://github.com/osexpert/ccs32clara-chademo/releases/tag/20260121-272
Thanks