osexpert/ccs32clara-chademo #28: Longood Technical discussion

Leslie48572733, 2025-09-17:
Hello Clara, it's great to meet you! I'm Leslie from Longood. We'd like to express our gratitude for your work on the open-source software for our adapter. We've tested your software and found its compatibility to be excellent. Our engineers are also interested in your software and would like to discuss some technical aspects with you. Please feel free to reach out if it's convenient for you. You can contact us at 
Boris @longoodint.com  
Leslie@longoodint.com 
or WhatsApp us at 
+86 13650432005 
 +86 13128122764.

osexpert, 2025-09-17:
Hi, Clara live for the most part in the https://github.com/uhi22/ccs32clara project, but now also living in other projects like this:-)
Its a nice adapter to hack and easy to flash, lowering the barrier for ccs and chademo hacking greatly:-)

uhi22, 2025-09-18:
I feel the same. Maybe it makes sense, to publish the current sources on a fresh github, and maybe someone from Japan who really needs this stuff jumps in. Personal use is the best motivation from my point of view. In my region, the chademo chargers are removed more and more, and I'm not sure whether I want to risk my CCS car for early development stage adapter experiments. But I'm interested to have a look in general and maybe contribute some thoughts.

Leslie48572733, 2025-09-21:
@ justingist Thank you for your information!Please let me know if you are a developer. If yes, please provide your address and phone number, and I will send you a sample of CHAdeMO to ccs2 adapter in order to create open source software.