osexpert/ccs32clara-chademo #32: Current limit

vogelkristofistvan, 2025-09-25:
I received a firmware version from the Longgood developers that limited the maximum charging current.
Would it be possible to limit the maximum charging current to 20 amps, and then increase it by pressing the button while charging?
For example: 25A, 35A, 50A, 70A, 100A, 125A, 150A, 200A

The car's BMS can still limit the charging current, this would only be the maximum the car can get.

Nissan Leaf battery cells are destroyed by overheating. A constant lower charging current results in more predictable heating.

osexpert, 2025-09-25:
Stop button during charging is already taken as stop button, I rather want to keep it like that..
It is possible to add a special mode (set right after power on) that limit to some fixed amps eg. 100.

osexpert, 2025-09-25:
Or if you build yourself, I think you can just change this line in main.cpp
    Param::SetInt(Param::MaxCurrent, 200);
and set the max current you want.

osexpert, 2025-09-25:
"Nissan Leaf battery cells are destroyed by overheating. A constant lower charging current results in more predictable heating."
At least for my car, 2018 Leaf 40kwh, in this summer, when in was 30 deg. Celsius, the battery temperature was pretty high all the time, and it did not increase at all during charging. The car typically requested around 30kw max, or worse. Limiting this to even lower, was the last idea in my mind:-)

vogelkristofistvan, 2025-09-25:
45 amps raises 10 degrees.

osexpert, 2025-09-25:
It would also be possible to add a special mode eg. 5 that change how the stop button works to how you suggest. But it seems like a very narrow use case, that would generally be usable only for a very few, so I will not involve myself into this. But being open source, its possible to make own special versions.

kivulitiger, 2026-01-30:
> Or if you build yourself, I think you can just change this line in main.cpp Param::SetInt(Param::MaxCurrent, 200); and set the max current you want.

I've done this on 20251217-232 and limited current to 60A. It seems to work fine but have only tested on an Alpitronic charger. 

[image]

kivulitiger, 2026-02-04:
It seems that Tritium and Alpitronic chargers respect the current limit, but Tesla V4 and Ocular chargers did not. I've uploaded the logs and it seems that 200A is still being taken from somewhere. In the case of Tesla V4 it is reporting as max 469V/500A available on CCS side but on Chademo side it is reporting as 469V/200A available. On Ocular same story but 500V/300A on CCS side and 500V/200A on Chademo. 

[SaveWindows4_02_2026_20-06-57_Occular.TXT](https://github.com/user-attachments/files/25071558/SaveWindows4_02_2026_20-06-57_Occular.TXT)
[SaveWindows4_02_2026_20-43-52_TeslaV4.TXT](https://github.com/user-attachments/files/25071559/SaveWindows4_02_2026_20-43-52_TeslaV4.TXT)

osexpert, 2026-08-21:
The latest release can be configured in the config editor and you can set any max amp you like, eg. 42:

[image]

closed 2026-08-21 by osexpert